UBNanosatLab/lfr-firmware #2: Modify txpwr commands to act on drain voltage instead of gate voltage

brianbezanson, 2019-01-15:
Subsequent to #1, the transmit power level should be controlled by changing the PA drain voltage.  Gate bias voltage during transmit will likely be a compile-time setting.  If we want to modify gate bias via commands during testing, we should probably create a different command for that.

iracigt, 2019-01-20:
With #4, this warrants further discussion on how the explicit SET_TX_PWR commands will act with regards to the configuration settings for the same. I'm in favor of just having them update the *volatile* configuration. It would be identical to doing a read-modify-write with the configuration, just quicker and easier for these commonly used settings.

brianbezanson, 2019-01-20:
Concur.  Since we seem to have settled on parameters being volatile by default and being committed to nonvolatile memory by a separate command, commands that set one parameter at a time should follow that same pattern.

If I were coming at this blindly, seeing a TX_POWER parameter in the config struct and a SET_TX_POWER command would lead me to expect (1) that the two representations refer to the same parameter, (2) if I send a config struct with TX_POWER = X, then send SET_TX_POWER to Y, then read back the config struct, I would see TX_POWER = Y, and (3) changes made by either means will revert upon reset, and changes made by either means will be made nonvolatile by the same command.

brianbezanson, 2019-01-31:
Pretty sure we just forgot to close this.  Can't recall what commit did it, but it's worked this way for a while now.

closed 2019-01-31 by brianbezanson