UBNanosatLab/lfr-firmware #4: Nonvolatile configuration

brianbezanson, 2019-01-16:
Allow configuration parameters to persist through resets and add a "factory reset" command to restore the configuration to a known state.  I see a few options for behavior:

1. Parameters are persistent by default.
2. Parameters are volatile by default, with a "use current values as defaults" command that makes the current parameter set persistent.
3. Parameters are volatile by default, with a per-parameter "set default value" command or command flag.

I think my favorite is 2.

brianbezanson, 2019-01-17:
Configurable parameters:

- [ ] Callsign
- [ ] TX gate bias 
- [ ] TX VDD
- [ ] PA current limit
- [ ] PA voltage ramp delay
- [ ] TX frequency
- [ ] RX frequency (later?)
- [ ] RF bit rate
- [ ] RF deviation (or roll that in with bit rate as a set of predefined combinations?)
- [ ] TCXO pull voltage
- [ ] Modulation flags (CRC, gaussian filtering)

brianbezanson, 2019-01-17:
In the configuration update command, include a config struct version number that increments whenever the structure changes.  Reject commands containing incompatible config structs.

iracigt, 2019-01-24:
Complete on settings branch! I left the frequency setting as one setting for both RX and TX. Modulation flags doesn't allow switching the Gaussian filter off. That can be made a separate issue if needed.

PR #18 (2019-01-24): Settings
Adds settings to LFR, along with the automated test script. Closes #4 and #8.
closed 2019-01-24 by iracigt