UBNanosatLab/lfr-firmware #11: Si4464 Hangs During TX (Rarely)

iracigt, 2019-01-20:
I think we might have hit another bullet point on the Si4464 errata sheet... Probably the one about packet handler locking up if told to TX while already doing so. If so, can be fixed with software interlock in MSP430. We should also look into a software TX timeout in addition to the rev2 hardware timeout (https://github.com/UBNanosatLab/lfr-hardware/issues/13) .

iracigt, 2019-01-20:
Reproduction is tricky, observed intermittently with amplifier sweep test

iracigt, 2019-02-10:
Seems to happen more often at higher speeds. I've reached out to silabs for help (https://www.silabs.com/community/wireless/proprietary/forum.topic.html/si4464_occasionalfr-0EGb), but in the meantime, I automatically recover by resetting the RFIC. Needs stress testing, but likely mitigated at this point.

brianbezanson, 2019-02-13:
Mitigated in 016408036cc2453434bc71429b5d4d1806f19b4a

closed 2019-02-13 by brianbezanson